Beckhoff-USA-Community/SPT-Libraries #1: How were the diagrams made?

runtimevic, 2022-10-27:
Hello,
How have the UML graphic diagrams been made to put them in the pdf documents?

MarkLewis-Beckhoff, 2022-10-27:
https://mermaid.live/

runtimevic, 2022-10-28:
> https://mermaid.live/

and how do you pass the classDiagram made in TwinCAT to mermaid?, 
or do you make it from scratch from mermaid?

nshiggins, 2022-10-28:
It is a separate tool not Beckhoff related.  The diagrams were made
manually.

nshiggins, 2022-10-28:
Not as far as I know.  Mermaid is its own markup language that just happens to have the ability to do generic class diagrams.  I chose to use it for a number of reasons, but mostly because I wanted to illustrate the relationships and not necessarily expose all of the locals, etc. in each function block.

closed 2022-10-28 by nshiggins

runtimevic, 2022-10-28:
> Not as far as I know. Mermaid is its own markup language that just happens to have the ability to do generic class diagrams. I chose to use it for a number of reasons, but mostly because I wanted to [...]

thanks,