Beckhoff-USA-Community/SPT-Libraries #4: Tracing: AddWithJson / AddEntryWithJson

stefanspiegel, 2022-12-16:
Hi all,

I have updated to SPT_Base_Types V3.0.1. The Function Block FB_BaseFB no calls a Method xxx.AddWithJson. This is not compiling.
Shouldn't it be xxx.Add**Entry**WithJson?

Thanks for your answer.

Best regards, Stefan

nshiggins, 2022-12-16:
Thanks Stefan I thought I fixed this yesterday--is it still an issue?

MarkLewis-Beckhoff, 2022-12-16:
Forgot the () ?

Sent from Nine<http://www.9folders.com/>

CAUTION: External Email!!

This email contains confidential information. If you have received it in error, you must not read, use, copy or pass on this e-mail or its attachments.
If you have received the e-mail in error, please inform me immediately by reply e-mail and then delete this e-mail from your system. Thank you

Diese E-Mail enthaelt vertrauliche Informationen. Sollten Sie sie irrtuemlich erhalten haben, duerfen Sie diese E-Mail oder ihre Anhaenge nicht lesen, verwenden, kopieren oder weitergeben.
Sollten Sie die Mail versehentlich erhalten haben, teilen Sie mir dies bitte umgehend per Antwort-E-Mail mit und loeschen Sie diese E-Mail dann aus Ihrem System. Vielen Dank

nshiggins, 2022-12-16:
No--the Trace() and TraceWithJson() methods in FB_BaseFB are just wrappers for the global tracing FB that is in SPT_Tracing library.   
 They call the global AddEntry() and AddEntryWithJson() and pass in the caller's FB instance name, etc.  The JSON stuff was a last minute addition and I changed the SPT_Tracing method name (was AddWithJson() --> AddEntryWithJson()) right before committing the libraries and didn't change the wrapper method.  Too excited to get it off my plate--my mistake.

closed 2022-12-16 by nshiggins