Beckhoff-USA-Community/SPT-Libraries #13: Motion Control Blended Moves

whitecider, 2023-05-10:
Hi,

I am trying to use FB_BasicAxis.RingBuffer_MCMoveAbsolute for a sequence of buffered moves. Is there a way to use a FB which IMPLIMENTS I_MotionSequence and have it blend the moves (ie)?
`_Axis.MC_MoveAbsolute.BufferMode := MC_BlendingNext`

I have used bActive of the MC_Move function blocks but that does not seem to be implemented here. 
````
Move1.Execute(TRUE);
If Move1.Active then
     Move2.Execute(TRUE);
END_IF
IF Move2.Active THEN
    Move1.Execute(FALSE);
    Move3.Execute(TRUE):
END_IF
````

nshiggins, 2023-05-10:
We could consider adding a property to the FB which allows you to plug in custom sequences like you describe (similar to HomeMethod, ResetMethod).  You are correct that all PTP moves are assumed to be MC_Aborting.  We talked about baking blending options into the method calls but it made the method signatures extra wordy.  I'll bring it up to the team

nshiggins, 2023-05-10:
Discussed with team and we will implement this in 2 ways:

- Add property ExternalSequence : I_MotionSequence where you can plug in an external sequence the same way HomeMethod and ResetMethod worked.  There will be an associated method to execute this.
- When 4026 is released we will have the ability to have optional method parameters.  We will modify the existing interface/FB to allow options to be passed to the underlying MC_ FBs.

Solution 1 I think we can work into v3.2 release, which is slowly being worked on currently.  Most of the team is assigned to paid projects so framework development comes in spurts.

runtimevic, 2023-05-11:
> Discussed with team and we will implement this in 2 ways:
>
> * Add property ExternalSequence : I_MotionSequence where you can plug in an external sequence the same way HomeMethod and ResetMethod [...]

Hello @nshiggins ,
This new feature of version 4026 sounds very good...
can you give a small example of optional method parameters?

nshiggins, 2023-05-11:
> > Discussed with team and we will implement this in 2 ways:
> > 
> > * Add property ExternalSequence : I_MotionSequence where you can plug in an external sequence the same way HomeMethod and ResetMethod worked.  There will be an associated method to execute this.
> > * When 4026 is released we will have the ability to have optional method parameters.  We will modify the existing interface/FB to allow options to be passed to the underlying MC_ FBs.
> > 
> > Solution 1 I think we can work into v3.2 release, which is slowly being worked on currently. Most of the team is assigned to paid projects so framework development comes in spurts.
> 
> Hello @nshiggins , This new feature of version 4026 sounds very good... can you give a small example of optional method parameters? You mean this? [image]

Yes, exactly.  More PLC updates here: https://www.beckhoff.com/en-us/support/webinars/the-new-twincat-3-1-build-4026-plc-updates.html

8:02 for this specific feature

closed 2023-06-02 by nshiggins

whitecider, 2025-05-15:
@nshiggins is this still being considered?