Beckhoff-USA-Community/SPT-Libraries #16: SPT Utilities could not open library Tc3_System

nicolagit, 2023-05-13:
Hi.

When I try to use SPT Utilities library version 3.0.2 in TwinCAT 3 build 4024.44 I get the error `Could not open library '#Tc3_System'. (Reason: The library 'Tc3_System, * (Beckhoff Automation GmbH)' has not been installed to the system.)`.

It's similar to [this issue](https://github.com/Beckhoff-USA-Community/SPT-Libraries/issues/6), but now in the library instead of the sample project.

Is there any solution for this?
Thanks

MarkLewis-Beckhoff, 2023-05-13:
please pull the Latest library version the references have been removed to this library they were not necessary any longer.

closed 2023-05-13 by MarkLewis-Beckhoff

nshiggins, 2023-05-13:
Sorry man I had the changes made to the Utilities library but somehow missed actually staging the files in the commit.  v3.0.3 is up now and should correct this Tc3_System issue (once and for all!)

nicolagit, 2023-05-13:
I confirm all sorted for me with latest commit.
Thanks all for the quick fix!