Beckhoff-USA-Community/SPT-Libraries #18: Cannot cast from/to PackML interfaces

calumsinclair, 2023-06-26:
Can you extend the existing packML interfaces so that we can run time cast them?

Use case example:
We want to implement I_SomeRandomComponent. We tend to inject our  I_SomeRandomComponent into the consuming class (a SubModule). We would like to cast the I_SomeRandomComponent to a I_Componentbase so that we can add it to the array. 

At the moment we pass our FB_SomeRandomComponent into our FB_Init twice, against different interfaces. One is of type I_SomeRandomComponent, the other against I_Componentbase it gets from the from the inherited FB_Componentbase.

MarkLewis-Beckhoff, 2023-06-29:
we have added the extends __SYSTEM.IQueryInterface to the I_BaseFB which all FB interfaces extend from you should be able to use the operator __QUERYINTERFACE at will. This should resolve the stated issue. Please confirm and I will close the issue.

calumsinclair, 2023-06-30:
Can confirm this is now added in, thanks.

closed 2023-06-30 by MarkLewis-Beckhoff