Beckhoff-USA-Community/SPT-Libraries #20: FB_Component_BasicAxis: JogFastVelocity/JogSlowVelocity never used

vanoo1983, 2023-07-28:
Hello

The ST_AxisBase_Config parameters JogFastVelocity/JogSlowVelocity are never used. So when I jog an axis, these parameters have no effect. Why do these parameters exist? Is this a bug or what is the idea behind it?

nshiggins, 2023-07-28:
This is a TODO that slipped through.  Will add to next release.  Original intent was to use NC parameters for jog speeds but currently uses Velocity property.

Changing this behavior might be a breaking change so it will likely require a library parameter to switch future behavior back to legacy.

MarkLewis-Beckhoff, 2023-08-01:
remove the unused variables from the structure now set jog velocities in the NC Parameters tab

MarkLewis-Beckhoff, 2023-08-01:
closing issue

closed 2023-08-01 by MarkLewis-Beckhoff