Beckhoff-USA-Community/SPT-Libraries #22: Sensor.Trace('My Sensor');  It only works the first time then it doesn't work more times...

runtimevic, 2023-07-28:
Hello,
Sensor.Trace('My Sensor');
It only works the first time then it doesn't work more times...
attached image...
[image]

nshiggins, 2023-07-28:
There is filtering in the trace logic which by default does not allow duplicate sequential messages.  This is to keep you from hammering the event logger in case you have a condition that is not a 'one-shot'.  You can override this with a library parameter.  You'll need to add the SPT Base Types library to your project in order to edit: 
[image]

closed 2023-07-28 by nshiggins

runtimevic, 2023-07-28:
Hello,
I have put the variable: ALLO_DUPLICATE_SEQUENTIAL_MESSAGES := FALSE;
and if I cyclically put the same message with: Sensor.Trace('Mi Sensor');
the message comes out in the output constantly without stopping...
Do I think it's wrong? Or how should I put it so that the same message doesn't come out repeatedly?

nshiggins, 2023-07-28:
I'm not clear on the sequence of events.  Originally with the default
library parameter setting you said the duplicate messages were being
filtered.  Now they are streaming constantly?  Did you modify the library
parameter from with the Library Repository window?

runtimevic, 2023-07-28:
> No tengo clara la secuencia de eventos. Originalmente, con la configuración de parámetros de biblioteca predeterminada, dijo que los mensajes duplicados se estaban filtrando. ¿Ahora están transmitiendo constantemente? ¿Modificó el parámetro de biblioteca desde la ventana Repositorio de biblioteca?
> […](#)

[image]
[image]
Now it doesn't filter the message, it comes out non-stop, that's what's happening to me now...,
It is cyclically set to be carried out and the message is constantly coming out

nshiggins, 2023-07-28:
> SPT Application Framework_Sensor_4]

Does it do the same after an activation/cold start?

runtimevic, 2023-07-29:
> ¿Hace lo mismo después de una activación/arranque en frío?
> […](#)

> ¿Hace lo mismo después de una activación/arranque en frío?
> […](#)

https://github.com/Beckhoff-USA-Community/SPT-Libraries/assets/53791834/c80937b9-9814-442a-965a-e96ba446c170

runtimevic, 2023-07-31:
could you look at it?

MarkLewis-Beckhoff, 2023-07-31:
Does it do the same after an activation/cold start?

This email contains confidential information. If you have received it in error, you must not read, use, copy or pass on this e-mail or its attachments.
If you have received the e-mail in error, please inform me immediately by reply e-mail and then delete this e-mail from your system. Thank you

Diese E-Mail enthaelt vertrauliche Informationen. Sollten Sie sie irrtuemlich erhalten haben, duerfen Sie diese E-Mail oder ihre Anhaenge nicht lesen, verwenden, kopieren oder weitergeben.
Sollten Sie die Mail versehentlich erhalten haben, teilen Sie mir dies bitte umgehend per Antwort-E-Mail mit und loeschen Sie diese E-Mail dann aus Ihrem System. Vielen Dank

MarkLewis-Beckhoff, 2023-07-31:
Does it do the same after an activation/cold start?

nshiggins, 2023-07-31:
Changes to constants need at least a PLC cold start to take effect.  It's working as designed on my end:

https://github.com/Beckhoff-USA-Community/SPT-Libraries/assets/18381949/7afd2734-36d0-4591-a1b2-220c7837b2aa

runtimevic, 2023-08-01:
https://github.com/Beckhoff-USA-Community/SPT-Libraries/assets/53791834/7c69515f-cf23-4542-993f-d702fbf718d9

watch the video, I hope you see it...
it's not working right...

MarkLewis-Beckhoff, 2023-08-01:
Hi Victor,
I will email you offline I would like to look at your code.

nshiggins, 2023-08-02:
Pushed SPT_Base_Types v3.2.3 today, which should fix this.  String comparison for detecting duplicate message was overcomplicated.