Beckhoff-USA-Community/SPT-Libraries #23: Space added in component name

MrKirkegaard, 2023-08-08:
The logic in FB_PackML_BaseModule.Initialize addes an extra space in the commponent names.

The Component name is "_**GL1.GL1.GMD1-YS1**_" according to ISO/IEC 81346, but in the SPT framework the component gets the name "**_GL1.GL1 .GMD1-YS1_**"

[image]

nshiggins, 2023-08-08:
Thanks will add to work items and update soon.  This could be a breaking change depending on existing HMI applications so will likely introduce a library parameter where user can specify delimiter.

MrKirkegaard, 2023-08-08:
Thank you.
Can the same type of name concatenation be added to the ipSubModules array, by means of a library parameter?

nshiggins, 2023-08-11:
This has been addressed and released in SPT_Base_Types v3.2.4

closed 2023-08-11 by nshiggins