Beckhoff-USA-Community/SPT-Libraries #26: SPT component base reference error

baseljaber, 2023-08-14:
the library **SPT component base** is referencing **tc3_system**, which does not exist, I guess it should reference **tc2_system**

Best regards
Basel Jaber

nshiggins, 2023-08-14:
Can you show me where this is the case?  This issue was addressed in v3.1.1:  https://beckhoff-usa-community.github.io/SPT-Libraries/SPT_Components/changelog.html

baseljaber, 2023-08-14:
You can see it when the library is referenced in Twincat: 

[image]

and when you open the dependencies file:

[image]

and it throws an error when compiling the project:

[image]

nshiggins, 2023-08-14:
SPT Component Types library has been deprecated.  That code now lives in the SPT Base Types library.

closed 2023-08-14 by nshiggins