Beckhoff-USA-Community/SPT-Libraries #30: Abort from Clearing not possible

MrKirkegaard, 2023-09-15:
The Parent response abort is not possible in state clearing. 
[image]
Both the SubModuleMonitor and ComponentMonitor do not allow the abort command defined in the PackML model.
[image]

KeithBaker123, 2023-09-15:
Thank you Mr Kirkegaard for your input to our repository.  We take in all issues and discuss them as a team.

On the issue that the SubmoduleMonitor and ComponentMonitor do not take the parentResponse in clearing and directly go to Aborting is done because we use Clearing to reset safety and other alarms. If we responded to them before being cleared, we would end up in a race condition.   So, we would usually have a timeout in Clearing and if the devices did not reset, we would send state engine to Aborting.

We feel that the framework follows PackML and our implementation of the Framework gives you the flexibility to do what you need.

MrKirkegaard, 2023-09-17:
Hi Keith

Thank you for the feedback. 

Is there a possibility that you can clarify this in the SPT Framework documentation, that ParrentResponseDefinition deliberately has no effect in state clearing?

Is it possible to add a Transition State Timeout function in FB_PackML_BaseModule?

I have had to Extended the FB_PackML_BaseModule to implement a Transition state timeout in all my Equipment Modules and the Unit.

closed 2023-09-17 by MrKirkegaard