Beckhoff-USA-Community/SPT-Libraries #35: Library SPT_Motion 3.2.5 FB_HomeRoutine_HardStop Build errors

AlplaAme, 2023-11-29:
Hello,

When using FB_HomeRoutine_HardStop(SPT_Motion 3.2.4 and 3.2.5), the following two Build errors occur:

[image]

Regards,
Rainer

cgrant11, 2023-11-30:
It looks like _Parameters is the wrong structure, should be ST_Home_Options2 not 3. 
What is FB_HomeRoutine_HardStop extending?

d3v1l6oy, 2023-11-30:
To allow the FB_BaseAxis to handle all types of homing you pass an interface of the function block that does the homing, to the FB_BasicAxis.Home Method.

Within FB_BasicAxis we cyclically call ipHomeMethod.CyclicLogic()
Within FB_BasicAxis we have ipHomeMethod : I_MotionSequence;
I_MotionSequence has an Execute Method and an Axis property.
Therefore, you can write your own homing routine that extends FB_CyclicFB and implements I_MotionSequence

Within the SPT Motion Control Library there are some examples
https://beckhoff-usa-community.github.io/SPT-Libraries/SPT_MotionControl/functionblocks.html#home-routines
 
 
Implementation Code below:

``` 
FUNCTION_BLOCK FB_EquipmentModule EXTENDS FB_PackML_BaseModule
VAR
    Axis            : FB_Component_BasicSlaveAxis;
    HomeRoutine     : FB_MyHomeMethod;
END_VAR
``` 
 
Method Initialize
```
Axis.HomeMethod := HomeRoutine;
```
 
Method Execute
```
CASE _CurrentMode OF
                E_PMLUnitMode.ePMLUnitMode_UserMode1: (*Homing Mode*)
                                                CASE SequenceState OF
                                                0:(*Init*);
                                                                IF NOT Axis.Homed THEN (*Typically the Calibration Flag in the NC - Axis.Status.Homed*)
                                                                                SequenceState := SequenceState +10;
                                                                ELSE
                                                                                SequenceState := 30;
                                                                END_IF
                                                10:(*Start Homing*); (*The method call will return TRUE if command accepted*)
                                                                IF Axis.Home() THEN
                                                                                SequenceState := SequenceState +10;
                                                                END_IF
                                                20:(*Wait*) (*Cyclic Logic will update the code in the background*)
                                                                IF Axis.Homed THEN
                                                                                SequenceState := SequenceState +10;
                                                                END_IF
                                                30:(*Move to start position*);
                                                                IF Axis.MoveAbsolute(Position := HOMEPOSITION, AbortPrevious:= TRUE) THEN
                                                                                SequenceState := SequenceState +10;
                                                                END_IF
                                                40: (*Complete*)
                                                                IF NOT Axis.Busy THEN                    (*Monitoring the Busy must be in a separate step from the method call to ensure a minimum number of PLC scans of Cyclic Logic*)
                                                                                _HomingRequired := FALSE;
                                                                                StateComplete();
                                                                END_IF
                                END_CASE           
```

nshiggins, 2024-03-01:
This has been fixed in SPT_Motion_Control v3.2.5.  Issue is advanced homing function blocks that we wrap use different options structs depending on the FB.  The ST_StepBlockLagBasedParameters was updated to include all relevant members of ST_Home_Options2/3, and are properly distributed in FB_HomeRoutine_HardStop now.

closed 2024-03-01 by nshiggins