Beckhoff-USA-Community/SPT-Libraries #39: Warnings with Tc3_JsonXML library Version 3.4.3.0

kaelev7, 2024-01-16:
Since the last TwinCAT update to version 4012.54 and the corresponding update of the `Tc3_JsonXML` library to version `3.4.3.0` I get the following warnings:

```
TcCOM TypeSystem: sub type 'ITcJsonDomParser3 ({86123BA3-7D31-4FBD-B2AA-90527A60B30F})' not found while importing data type 'Tc3_JsonXml.FB_JsonDomParserBase'! Data type will be ignored				
TcCOM TypeSystem: base type 'SPT_BaseTypes.FB_TraceLog' not found importing data type 'Tracing.GlobalTraceLog'! Data type will be ignored				
TcCOM TypeSystem: error while importing data type 'SPT_BaseTypes.FB_TraceLog'! Data type will be ignored				
TcCOM TypeSystem: error while importing data type 'Tc3_JsonXml.FB_JsonDomParserBase'! Data type will be ignored				
TcCOM TypeSystem: extends type 'Tc3_JsonXml.FB_JsonDomParserBase' not found importing data type 'Tc3_JsonXml.FB_JsonDomParser'! Data type will be ignored				
TcCOM TypeSystem: sub type 'Tc3_JsonXml.FB_JsonDomParser' not found while importing data type 'SPT_BaseTypes.FB_TraceLog'! Data type will be ignored				
TcCOM TypeSystem: subitem 'ipDom3' not found importing data type 'Tc3_JsonXml.FB_JsonDomParserBase'! Data type will be ignored				
TcCOM TypeSystem: subitem 'JsonParser' not found importing data type 'SPT_BaseTypes.FB_TraceLog'! Data type will be ignored		
```
	
After setting the library version back to use version `3.3.19.0`	the warnings disappear.   
Since I don't use the TraceLog and not using any of the `Tc3_JsonXML` library in my own code, I don't mind setting it back to an old version.

nshiggins, 2024-01-16:
Some of us are seeing the same thing, but others are not.  Strange.  It seems that if you add a reference to the JsonXML library to your project, the warnings disappear.  Oddly enough you can then remove the reference and leave the BaseTypes reference to the library as published (always newest), the warnings remain gone.

kaelev7, 2024-01-17:
With the new SPT_Base_Types version 3.2.9.1 I get the same behavior.
I need to add the Tc3_JsonXML library version 3.3.19.0 as a reference to the project.

But I only get this warnings after starting the XAE. After a rebuild of the solution, I don't get any warning. no matter if I use the old version of the JsonXML library or the new version.

closed 2024-03-01 by nshiggins