Beckhoff-USA-Community/SPT-Libraries #41: TC3_System

giordanoceccarini, 2024-01-31:
I can not found Tc3_System library

MarkLewis-Beckhoff, 2024-01-31:
all references we could find to that library were removed many versions ago.

are you running the latest build of the libraries?

which library have you found this error in?

giordanoceccarini, 2024-01-31:
> all references we could find to that library were removed many versions ago.
> 
> 
> 
> are you running the latest build of the libraries?
> 
> 
> 
> which library have you found this error in?

Hi. I find version 3.0.1, but now I found 3.0.1 that not had tc3-system reference. I trini i Am ok

albertgoodwin, 2024-06-25:
[image]

i still see TC3_system as a dependency in NC_Homing, which is causing build fails. What am i missing?