Beckhoff-USA-Community/SPT-Libraries #43: SPT Motion Control - ST_AxisStatus - DriveDeviceError

vanoo1983, 2024-02-21:
If in a axis the status DriveDeviceError is set, the FB_BasicAxis has no error. The Error of FB_BasicAxis is FALSE and the ErrorID is 0. 
But if DriveDeviceError ist set, there must be called MC_Reset to get out of this error. 
The Reset method will only be called if a Componet has an error. If FB_BasicAxis has no error also FB_Component_BasicAxis has no error. So the Reset method can not be called.

How is your idea to get out of this error?

Regards

Silvano

nshiggins, 2024-02-21:
Thanks and good point--this will be fixed in the next release

nshiggins, 2024-03-01:
This has been fixed.  DriveDeviceError now should throw an error.  This was added in motion control library v3.2.5.  Components have been updated to reflect this error as well, v3.2.7.  

We also added [parameter](https://beckhoff-usa-community.github.io/SPT-Libraries/SPT_Base_Types/globals.html#parameters_packml_base) to base types v3.2.10 that allows you to reset all components regardless of error flag.

In the next major release, this will be the default behavior.  There will be other breaking changes in the next major release (4.x) so we figured best way to not break existing applications would be the parameter.

closed 2024-03-01 by nshiggins