Beckhoff-USA-Community/SPT-Libraries #49: XPlanar TrackMove Method

CopeWalkerBeckhoff, 2024-04-11:
IMO:

It would make more sense to rename the existing TrackMove method "TrackMoveXY" and create a new TrackMove method with the one dimensional track position as an input.

MarcWilkinson, 2024-04-16:
Thank you for the feedback.  I originally used the X and Y input arguments as convenience item to save the user the step of calculating the dimensional track position, but I can see use cases where having that ability would be beneficial.  I will look into getting this implemented in the next release.

CopeWalkerBeckhoff, 2024-04-16:
Thanks!

In my case I'm indexing a mover along path for an assembly process.  I set it up so that a track would be in the desired mover path, and you move a mover to that track and then index it along the track, so if the customer moves the tooling in the future (especially if you move it to index along Y instead of X for example), you just need to move the track vs re-write the indexing code.

MarcWilkinson, 2024-05-24:
This will become available in a future release.  Closing out the issue.

closed 2024-05-24 by MarcWilkinson