Beckhoff-USA-Community/SPT-Libraries #55: Method ModeChanged() is called after the LastState assignment statement.

ulysses936, 2024-05-12:
The ModeChanged() method is called when the PackML mode is changed, but currently there is no way to determine the LastMode at the time of the method call because the LastMode is assigned to the current Mode before the method is called.

If the mode before the change of mode is available, it is possible to apply different logic for different modes. For example, if I only want to trigger a logic when switching from Auto to Manual, but not when switching from Invalid to Auto mode at machine startup, LastMode is necessary.

I think it would be better to call the ModeChanged() method before LastMode is reassigned, but I'm not sure if that would have any other implications.

[image]

SimmelFlo, 2024-05-12:
I can see your use-case. 
We will review the request and if there are not negative side-effects we will roll it into the next release.

nshiggins, 2024-05-12:
This would be nearly impossible to regression test.  I like the idea
though.  Maybe another library parameter to allow the timing of the call to
differ?

ulysses936, 2024-05-13:
I moved the ModeChanged method call before LastMode is assigned and generated a new library file, no problems encountered so far, I think it's because the ModeChanged method isn't used anywhere else in the library, only the user of the library overrides the method to add their own logic.

By the way, when I referenced the library to my newly exported library, I noticed that the Tracing related EventClass was not automatically added, is this something that needs to be set somewhere?

FlorianSiBeckhoff, 2024-08-21:
By changing the call order, it is still possible to set the 3 state variables to initial values by the user right at the beginning of the call of ModeChanged/StateChanged. 

Change is staged for release.

closed 2024-08-21 by FlorianSiBeckhoff