Beckhoff-USA-Community/SPT-Libraries #74: FB_Component_NciBase gives 'interpreter error' at initializion with twincat 4026

micfonon, 2024-09-10:
I migrated to TC 4026, almost everything works fine, but Nci does not initialize and raises an error:

> Error		10/09/2024 09:55:33 998 ms | AlarmCleared | 'MAIN.Nci': Nci - Interpreter Errror! (No error id found) (INITFAILED)			

This is the Nci interpreter state, it is initialy in state 0 (INIT_FAILED), but can't get out of it because FB_Component_NciBase raises an alarm and sets _Error to TRUE;

If (in XAE) you manually build a group and reset the NC module, it works.

Looks like NC's behaviour in TC 4026 changes slightly?

FlorianSiBeckhoff, 2024-09-10:
I have tested the function of the library with version 4026.10 XAE & XAR. 
The issue described above was not present. Can you please share your installed versions with us? 
(tcpkg list -i Nci) & (tcpkg list -i XAR)

The alarm was cleared in the description above. Which indicates that Nci.Reset() has been called. 
If you then see an error after, means that the Interpreter is still in fault. Which Fault ID is shown in the Interpreter GUI?

micfonon, 2024-09-12:
The ToPlc.ItpState stays at zero, and the ToPlc.ErrorCode is 0, also after Nci.Reset();

i added the tpckg list below, and also a (very simple) example code and steps to reproduce my problem:

**tcpkg list -i Nci on PLC:**
TF5100.NCI.XAR 1.0.0
TwinCAT.XAR.NCI.Classic 3.1.5006
TwinCAT.XAR.NCI.GST 1.0.9

**tcpkg list -i XAR on PLC:**
TF3300.ScopeServer.XAR 34.48.18
TF5000.NCPTP.XAR 1.0.0
TF5100.NCI.XAR 1.0.0
TwinCAT.Standard.XAR 4026.10.0
TwinCAT.XAR.Base 2.2.33
TwinCAT.XAR.CxDevice 1.1.6
TwinCAT.XAR.DriversBase 3.10.0
TwinCAT.XAR.ESB 1.4.3
TwinCAT.XAR.EtherCATSlave 2.1.3
TwinCAT.XAR.EtherNetIP 3.0.9
TwinCAT.XAR.NCI.Classic 3.1.5006
TwinCAT.XAR.NCI.GST 1.0.9
TwinCAT.XAR.NCPTP 3.1.5016
TwinCAT.XAR.NCPTP.UMRT 3.1.5012
TwinCAT.XAR.NdisDriver 2.1.7
TwinCAT.XAR.PLC 2.0.38
TwinCAT.XAR.PnP 1.3.3
TwinCAT.XAR.Profinet 2.2.16
TwinCAT.XAR.Realtime 2.0.63
TwinCAT.XAR.ReportingServer 1.2.29
TwinCAT.XAR.ScopeServer 4.48.44
TwinCAT.XAR.SystemServiceLib 1.1.0
TwinCAT.XAR.USB 1.3.3

**my very small test application:**
PROGRAM MAIN
VAR
	Nci : FB_Component_NciBase := (Name := 'Nci', Override := 1_000_000);
	Seq :INT := 0;
	InterpreterState : UINT;
	ResetCmd : BOOL;
END_VAR
----------------------
Nci.CyclicLogic();
InterpreterState := Nci.InterpreterState;
IF NOT Nci.InitComplete THEN
	RETURN;	
END_IF

CASE Seq OF
0:
	IF ResetCmd THEN
		Seq := 5;
	END_IF
5:
	IF Nci.Reset() THEN
		Seq := 10;
	END_IF
	
10:
    IF NOT Nci.Busy AND NOT Nci.Error THEN
        Seq := 20;
    END_IF
END_CASE;

**When I run the test application:**
1. power on
2. activate / restart run mode
3. login / create port
4. start
-> InterpreterState stays 0, and I immediately get an error:
Severity	Code	Description	Project	File	Line	Suppression State
Error		12/09/2024 10:31:48 862 ms | AlarmRaised | 'MAIN.Nci': Nci - Interpreter Errror! (No error id found) (INITFAILED)				
5. force reset
-> InterpreterState still 0, alarm is cleared and raised again: 

Severity	Code	Description	Project	File	Line	Suppression State
Error		12/09/2024 10:31:48 862 ms | AlarmRaised | 'MAIN.Nci': Nci - Interpreter Errror! (No error id found) (INITFAILED)		
Error		12/09/2024 10:33:05 682 ms | AlarmCleared | 'MAIN.Nci': Nci - Interpreter Errror! (No error id found) (INITFAILED)				
Error		12/09/2024 10:33:05 692 ms | AlarmRaised | 'MAIN.Nci': Nci - Interpreter Errror!

FlorianSiBeckhoff, 2024-09-12:
Please follow all steps of the quickstart.
[image]

Once the Axis-Ids are added to the sample program it works as intended.

closed 2024-09-12 by FlorianSiBeckhoff