Beckhoff-USA-Community/SPT-Libraries #78: No longer able to add as a PLC Library Repository on 4026.11

hersh-icon, 2024-10-15:
### Issue
When I upgraded from 4026.10 to 4026.11 I had to re-add the SPT Libraries as a PLC library Repository. I followed documentation verbatim (which as an aside is quite good :)) but I'm encountering the following error indicating that the path is not the root of a repository.
[image]
I've tried the same steps multiple times in multiple projects opening them in the XAE shell and Visual Studio 2022. One colleague experienced the same issue and after hours of re-trying the same steps, it finally worked (she could not pinpoint what she did to break it through). Another colleague had no problems with re-adding the library repository location... I had this issue when I upgraded to 4026.10 but after a few tries and some PC reboots it was resolved. Now, on 4026.11, after hours of trying basic troubleshooting steps I'm still stuck.

### System Configuration
* I've tried checking out the following commits to see if it will finally detect the directory as a valid root of repository:
  * 6fceb55 - Fixing naming to match
  * 1cc07c2 - Publishing GitHub Pages  ***NO_CI***
  * bf6313f - Merge pull request #72 from Beckhoff-USA-Community/SPT_MotionV3.2.7
  * d44fd22 - Merge branch 'main' of https://github.com/Beckhoff-USA-Community/SPT-Libraries
* Not sure if it it's helpful but here are my XAE package versions.
```
❯ tcpkg resolve TwinCAT.Standard.XAE
TcPkg 2.1.90

TwinCAT.Prep 1.0.16
TwinCAT.Standard.XAE 4026.11.0
TwinCAT.XAE.AdsApi 1.1.3
TwinCAT.XAE.Base 1.2.0
TwinCAT.XAE.DALI 1.0.0
TwinCAT.XAE.Debugger 2.7.8
TwinCAT.XAE.Debugger.TCXAESHELL64 2.7.8
TwinCAT.XAE.Debugger.VS2022 2.7.8
TwinCAT.XAE.DocDataProvider 1.1.4
TwinCAT.XAE.DocDataProvider.TCXAESHELL64 1.1.4
TwinCAT.XAE.DocDataProvider.VS2022 1.1.4
TwinCAT.XAE.EL6201 1.0.5
TwinCAT.XAE.EL6692Master 1.0.5
TwinCAT.XAE.EL6692Slave 1.0.4
TwinCAT.XAE.EL6695 1.0.8
TwinCAT.XAE.EtherCATConfig 2.0.32
TwinCAT.XAE.EtherCATSlave 2.0.5
TwinCAT.XAE.EtherCATTopology 1.2.0
TwinCAT.XAE.Eventlogger 1.4.4
TwinCAT.XAE.Eventlogger.TCXAESHELL64 1.4.4
TwinCAT.XAE.Eventlogger.VS2022 1.4.4
TwinCAT.XAE.FieldbussesBase 1.8.0
TwinCAT.XAE.HART 1.0.1
TwinCAT.XAE.Integration 2.8.0
TwinCAT.XAE.Integration.TCXAESHELL64 2.8.0
TwinCAT.XAE.Integration.VS2022 2.8.0
TwinCAT.XAE.IOLink 1.1.18
TwinCAT.XAE.PLC 5.4.4
TwinCAT.XAE.PLC.Lib.Tc2_Drive 3.3.13
TwinCAT.XAE.PLC.Lib.Tc2_EtherCAT 3.5.1
TwinCAT.XAE.PLC.Lib.Tc2_EthernetIP 1.1.2
TwinCAT.XAE.PLC.Lib.Tc2_IoFunctions 3.4.4
TwinCAT.XAE.PLC.Lib.Tc2_Math 3.4.4
TwinCAT.XAE.PLC.Lib.Tc2_MC2 3.3.65
TwinCAT.XAE.PLC.Lib.Tc2_MC2_Drive 3.3.43
TwinCAT.XAE.PLC.Lib.Tc2_ProfinetDiag 1.5.2
TwinCAT.XAE.PLC.Lib.Tc2_Standard 3.4.5
TwinCAT.XAE.PLC.Lib.Tc2_System 3.6.4
TwinCAT.XAE.PLC.Lib.Tc2_Utilities 3.8.2
TwinCAT.XAE.PLC.Lib.Tc3_DriveMotionControl 3.0.13
TwinCAT.XAE.PLC.Lib.Tc3_DynamicMemory 1.1.2
TwinCAT.XAE.PLC.Lib.Tc3_EventLogger 3.3.10
TwinCAT.XAE.PLC.Lib.Tc3_IoLink 3.4.4
TwinCAT.XAE.PLC.Lib.Tc3_IPCDiag 1.2.1
TwinCAT.XAE.PLC.Lib.Tc3_JsonXml 3.4.5
TwinCAT.XAE.PLC.Lib.Tc3_Module 3.4.5
TwinCAT.XAE.PLC.RM.4026.11 5.4.4
TwinCAT.XAE.PLC.TCXAESHELL64 5.4.4
TwinCAT.XAE.PLC.VS2022 5.4.4
TwinCAT.XAE.ProjectCompare 2.0.17
TwinCAT.XAE.ProjectCompare.TCXAESHELL64 2.0.17
TwinCAT.XAE.ProjectCompare.VS2022 2.0.17
TwinCAT.XAE.SafetyEditor 1.2.13
TwinCAT.XAE.SafetyEditor.TCXAESHELL64 1.2.13
TwinCAT.XAE.SafetyEditor.VS2022 1.2.13
TwinCAT.XAE.StandaloneTypesystem_1_16 1.16.25
TwinCAT.XAE.SystemManagerBase 2.4.46
TwinCAT.XAE.SystemManagerBase.RM.4026.11 2.4.46
TwinCAT.XAE.SystemManagerInterops 2.4.1
TwinCAT.XAE.SystemManagerVS 2.4.46
TwinCAT.XAE.SystemManagerVS.RM.4026.11 2.4.46
TwinCAT.XAE.SystemManagerVS.TCXAESHELL64 2.4.46
TwinCAT.XAE.SystemManagerVS.VS2022 2.4.46
TwinCAT.XAE.TMC 1.5.6
TwinCAT.XAE.TMC.TCXAESHELL64 1.5.6
TwinCAT.XAE.TMC.VS2022 1.5.6
TwinCAT.XAR.Base 2.2.34
TwinCAT.XAR.CxDevice 1.1.6
vcredist100.Beckhoff 10.0.40219.325
vcredist140.Beckhoff 14.38.33130
```

* Using stable feed only
```
❯ tcpkg source list
TcPkg 2.1.90

Beckhoff Stable Feed - https://public.tcpkg.beckhoff-cloud.com/api/v1/feeds/stable/ (Authenticated) | Priority - 1 | Bypass Proxy - False | Include Prereleases - False
Beckhoff Testing Feed - https://public.tcpkg.beckhoff-cloud.com/api/v1/feeds/testing [DISABLED] (Authenticated) | Priority - 2 | Bypass Proxy - False | Include Prereleases - False
```

barry-baker, 2024-10-15:
According to Beckhoff USA support, this is an issue with 4026.11 and is not specific to the SPT Library.  They recommend downgrading to a previous TC build.

99jj234, 2024-10-15:
Any path to making the libraries compatible with the new version?

barry-baker, 2024-10-15:
I was told the product manager would be notified about the issue but not given any further information.

nshiggins, 2024-10-15:
Some have had success deleting the cache file from the library repository
directory.

hersh-icon, 2024-10-15:
> Some have had success deleting the cache file from the library repository directory.
> […](#)

Thanks @nshiggins that worked!

brazinga, 2024-12-26:
Tc  4026.12 also has the same issue, somehow it resolved by deleting cache file

thoft99, 2025-01-08:
> Tc 4026.12 also has the same issue, somehow it resolved by deleting cache file

Can please elaborate on where this cache file lives?

Yura6969, 2025-01-10:
Read the manual. Everything works and adds without problems.
https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/4189425291.html&id=

Yura6969, 2025-01-10:
For new repositories you can only use empty directories or a valid existing repository.

mcclureTC, 2025-05-08:
cache files removed from repo with commit: a210ed258462bae9f7c09ab2d95ae998d83c7054, PR#97

closed 2025-05-08 by mcclureTC