Beckhoff-USA-Community/SPT-Libraries #79: Add the ability to change an alarm severity

SamFisco-Maskine, 2024-10-23:
When adding alarms to the Type System, they have pre-defined severity. When using the F_CreateAllEventsInClass, the severity is not adjustable. Ideally, some function or method as apart of the event handling library that takes in the original alarm FB and changes the severity to whatever the set severity is.

mcclureTC, 2024-10-23:
What is the use case for this? 

When starting a project, I create a tmc file for each component that has one alarm for each severity type (Verbose, Info, Warning, Error, Critical). This allows me to raise an alarm for each type in order to achieve the desired AlarmResponse during initial development, and the information about the alarm is provided by the String inputs of F_RaiseAlarmWithStringParameters. Later when ready to add specific alarms, I do so with the alarm severity set so that the desired AlarmResponse is reached.

SamFisco-Maskine, 2024-10-23:
A good use case could be for assigning different severities to analog input LowLow/Low/High/HighHigh limits. You would have the same FB for each analog input but can change the severity for each alarm condition to take a different action as needed. This way, you wouldn't need to create an alarm for each severity during the initial development and would only need to create the specific alarms

Bjerg8600, 2024-10-27:
SamFrisco do you need to change the severity or do you really only need to change the reaction to the alarm on the parent module? 
_ParentFaultResponseDefinitions : ARRAY[0..4] OF E_AlarmResponse := [   E_AlarmResponse.NoResponse, 
                                                                            E_AlarmResponse.NoResponse, 
                                                                            E_AlarmResponse.NoResponse, 
                                                                            E_AlarmResponse.Abort_ImmediateError, 
                                                                            E_AlarmResponse.Abort_ImmediateError];

SamFisco-Maskine, 2024-10-28:
Using the reaction to the alarm on the parent module would work but having the ability to change, or assign the severity to the alarm in the initialization step, would be preferred. 

For example, an Analog Input FB has LowLow/HighHigh limits as Error severity and Low/High limits as Warning severity. If I need to trigger an immediate shutdown of my system if the HighHigh is reached, I cannot check the severity of the alarm because of the LowLow limit possibly shutting of the system. Ideally, I could just set the HighHigh severity to Critical for that specific function block instance and abort the system immediately if any alarm is of Critical severity. 

If this breaks best practices or is not implementable within the SPT libraries then I can determine some work-around or wrapper for the severity.

mcclureTC, 2024-10-29:
@SamFisco-Maskine, I have added a sample project that includes an example of changing alarm severity at run-time.

https://github.com/Beckhoff-USA-Community/SPT_Samples/tree/main/TwinCAT%20PLC%20Projects/SPT_Alarms

closed 2025-05-06 by mcclureTC