Beckhoff-USA-Community/SPT-Libraries #87: Slave axis

giordanoceccarini, 2024-12-12:
I am thinking to create a FB (FB_GearCamSlave) that can manage Gear, Cam and Superimposed also. I think this it is necessary for many application. In my next applications I would like to use SPT Framework because I think it is the best approach for modern application, by the way, actually, SPT Framework is missing some "Objects" like describe.
Giordano

giordanoceccarini, 2024-12-12:
Did you think to a FB_GearCamSlave as in my mind to add in your SPT Framework?

mcclureTC, 2024-12-12:
It has been my experience that it is best to use virtual axes between a real axis and a Cam table. The existing FB_Component_BasicSlaveAxis uses the Tc2_MC2.MC_GearInMultiMaster which allows for up to 4 masters to be defined for an axis (much easier than SuperImposed). If one of these masters is a Virtual that is of type FB_Component_CamSlaveAxis then the management of the real axis becomes simplified. I am currently working on some sample code for these situations.

giordanoceccarini, 2024-12-12:
Thank you. Very interesting what you say. I think an example is very useful

mcclureTC, 2025-01-15:
@giordanoceccarini  there is now a sample available: https://github.com/Beckhoff-USA-Community/SPT_Samples/tree/main/TwinCAT%20PLC%20Projects/SPT_MultiMaster

giordanoceccarini, 2025-01-15:
Thank you. I am very excited to start this new avventure. I am using TwinCAT and Beckhoff product since 2012

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

giordanoceccarini, 2025-01-16:
Download and check every solution.
Just to keep practice with every single component I prefere to test them singolary.
If I want to test an axis (basic) just create an istance (MAIN) and call axis.cyclicLogic()?

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

giordanoceccarini, 2025-01-20:
FB_Component_BasicSlaveAxis uses MC_GearInMultiMaster which supports coupling to an axis that is already moving.
I was testing and it is TRUE only if flag AdvancedSlaveDynamics is TRUE.
By default flag AdvancedSlaveDynamics is FALSE and no Acceleration is considered (JUMP to master velo without ramp).
In your component how can I set TRUE flag AdvancedSlaveDynamics? For my test I did not use SPT Framework

mcclureTC, 2025-01-21:
Currently the Options input structure of the MC_GearInMultimaster is not accessible through the component.

giordanoceccarini, 2025-01-21:
It means that now it is not possible from slave to link a master at speed

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

giordanoceccarini, 2025-10-18:
> Currently the Options input structure of the MC_GearInMultimaster is not accessible through the component.

Hi, any news about possibility to access at the Option Input structure of the MC_GearInMultimaster?
Personally it is very important because I need gear one or more slaves to Master already in moving. I know I can use Cam but with GearInMultimaster it is so easy

giordanoceccarini, 2025-10-21:
> the basic slave has always been a gearin multi https://beckhoff-usa-community.github.io/SPT-Libraries/SPT_MotionControl/interfaces.html the interface shows 4 available masters.

hi. I know slave axis has 4 master available, but in case master is already moving how can i gear slave to it? Master can be geared but with jump.

Inviato da Outlook per iOS<https://aka.ms/o0ukef>