Beckhoff-USA-Community/SPT-Libraries #89: ErrorID

giordanoceccarini, 2025-01-16:
I create an instance of FB_Component_BasicAxis on MAIN.
Call .CyclicLogic() on MAIN.
I Create a machine state to run simple test.
1. Enable
2. MoveVelocity
3. Stop
4. Disable

Everything is fine.
Now I generated 2 error condition whike running:
1. Open STO on drive
2. Disable axis with Method Disable()

Scenario 1 give me TwinCAT error 16992 and also your component give me error 16992 as Logged events, but why ErrorID in Component give 0
Scenario 2 give me TwinCAT error 18000 and also your component give me error 18000 as Logged events and now ErrorID in Component give 18000

Is it correct? is it what you expect? I think also scenario 1 should give me 16992
Regards
Giordano

mcclureTC, 2025-01-16:
Opening the STO or Disabling before stopping the axis will create error 16992, this is expected. To prevent these error messages, you can first stop the axis using the method Stop().

4260 | 16992 | Monitoring | "Controller enable" Controller enable for the axis or for a coupled slave axis is not present (see axis interface PlcToNc). This error occurs if the controller enable is withdrawn while an axis or a group of axes (also a master/slave group) is being actively positioned. The error also occurs if a PTP axis or a coupled slave axis is started without controller enable.

16992: https://infosys.beckhoff.com/content/1033/tcncerrcode2/513129739.html?id=2564349323816727852

giordanoceccarini, 2025-01-16:
I know the error, my question was related to your component… why 18000 error has been found on errorid and why 16992 not

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

mcclureTC, 2025-01-16:
The error 16992 is working properly for me:

[image]

The error 18000 is related to hardware and I am not able to test this. It might be better contact your local support.

giordanoceccarini, 2025-01-17:
I see. Very very strange. May I have your example? I use 4024.64

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

image.png (view on web)<https://github.com/user-attachments/assets/3b8221cf-a91c-4e41-b355-be3b02e2f0a8>

giordanoceccarini, 2025-01-17:
[cid:0eaeaca9-c6d7-4c5c-91a1-d76f48b6bc62][cid:0c460d07-74da-42a2-af43-46213702058f]

As you can see, in my side no error

giordanoceccarini, 2025-01-17:
I have another question.
In case I have a master running at speed and slave axis stop. May I start slave with Gear? Normally I use MC_GearInVelo for this

mcclureTC, 2025-01-17:
Internally the FB_Component_BasicSlaveAxis uses MC_GearInMultiMaster which supports coupling to an axis that is already moving.

giordanoceccarini, 2025-01-17:
Thanks. I will try again and let you know because this morning I heared a “bump” when start gear and master was in run

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

giordanoceccarini, 2025-01-17:
Did you see my pictures about error 16992?

Inviato da Outlook per iOS<https://aka.ms/o0ukef>

giordanoceccarini, 2025-01-18:
Yes, you are right.

By default flag AdvancedSlaveDynamics is FALSE and no Acceleration is considered (JUMP to master velo without ramp).
In your component how can I set TRUE flag AdvancedSlaveDynamics? For my test I did not use SPT Framework