Beckhoff-USA-Community/SPT-Libraries #90: Cannot build because of Tc3_PackML_V2

ptaenzer, 2025-01-23:
I just cloned the repo a few hours ago and i went to make a component following the video tutorials. When I want to build the solution I get these four errors. 

[image]

It seems to be an error with line 7 of Parameters_PackML_Base

[image]

mcclureTC, 2025-01-23:
The library "Tc3_PackML_V2" must be added to your project references.

[image]

ptaenzer, 2025-01-23:
That line is yellow I think for me

[image]

ptaenzer, 2025-01-23:
> The library "Tc3_PackML_V2" must be added to your project references.
> 
> [image]

When I hover over that line it says Tc3_PackML_V2 library is not available

mcclureTC, 2025-01-24:
Please use the following PowerShell command to download and install the Tc3_PackML_V2 library:

TcPkg install TwinCAT.XAE.PLC.Lib.Tc3_PackML_V2 -y

Regarding your image with the yellow icon, this is showing the list of dependencies for the SPT Base Type library, and the yellow icon is indicating a problem with the library.

In order for you to directly reference part of the library in your code, you will have to additionally add the library to your project after completing the above install.

ptaenzer, 2025-01-24:
> Please use the following PowerShell command to download and install the Tc3_PackML_V2 library:
>
> TcPkg install TwinCAT.XAE.PLC.Lib.Tc3_PackML_V2 -y
>
> Regarding your image with the yellow icon, this is [...]

Good Morning,

Thank you for all of this help. I just want to make sure I fully understand. When I run that install should I be doing it inside of the folder for twincat on my machine or within the spt folder I created when I cloned this repo or just on the pc? Then once that is done I need to add it to the references within my project? 

Thanks again for your help

closed 2025-01-24 by mcclureTC