Beckhoff-USA-Community/SPT-Libraries #95: TwinCAT 3.1.4026 cannot add path for libraries

runtimevic, 2025-03-08:
Hello,
TwinCAT 3.1.4026 can

[image]

[image]

not add path for libraries

giordanoceccarini, 2025-03-10:
Same problem for me.

barry-baker, 2025-03-10:
https://github.com/Beckhoff-USA-Community/SPT-Libraries/issues/78

cgrant11, 2025-03-10:
Deleting the cache file fixed it for me

runtimevic, 2025-03-10:
> Deleting the cache file fixed it for me

Can you put an image or how to delete the cache?

runtimevic, 2025-03-10:
> [#78](https://github.com/Beckhoff-USA-Community/SPT-Libraries/issues/78)

Can you put an image or how to delete the cache?

runtimevic, 2025-03-10:
Deleting file named cache in /SPT-Libraries/Library Repository/ fix the problem.