Beckhoff-USA-Community/SPT-Libraries #103: Issue when adding repository location

Olli-M, 2025-04-15:
I am having issues when adding the library. As far as I know everything should be as it is, I have attempted cloning through tortoisegit and twincat, as well and all three depths in case it has changed since writing the intro. That said, it is not working, and I have no idea why. I have attached a screenshot of the error message below:

[image]

Olli-M, 2025-04-15:
Thanks, that fixed it, maybe update the readme and getting started guide to include that step for those of us (me) who don't know any better?

cgrant11, 2025-04-15:
sebvc pull request should fix the issue, but it needs to be reviewed and merged by someone at Beckhoff

mcclureTC, 2025-05-08:
cache files removed from repo with commit: a210ed258462bae9f7c09ab2d95ae998d83c7054, PR#97

closed 2025-05-08 by mcclureTC