Beckhoff-USA-Community/SPT-Libraries #109: A question from a beginner

NothingWorryGee, 2025-05-14:
FB_CyclicFB is required to implement the properties Busy, Error, and ErrorID from the interface I_CyclicFB. However, FB_CyclicFB also inherits these same properties (Busy, Error, and ErrorID) from its parent class FB_BaseFB, which creates a puzzling scenario. How should this be interpreted? Would it be more appropriate to declare the Busy, Error, and ErrorID properties in FB_BaseFB as private?

mcclureTC, 2025-05-14:
First, I would like welcome you and make sure you are aware of the Samples and Documentation.
Samples: https://github.com/Beckhoff-USA-Community/SPT_Samples
Documentation: https://beckhoff-usa-community.github.io/SPT-Libraries/

Both FB_ComponentBase and FB_PackML_BaseModule; which are the foundations of the framework, extend FB_CyclicFB. Because of the inheritance, all 3 of these have access to the Busy, Error, and ErrorID of their parent's class (FB_BaseFB). 

There is no requirement for you to use FB_CyclicFB to use these properties. If you only want to use these properties, then you can extend FB_BaseFB. However, it is more likely that you should be extending FB_CyclicFB (or FB_ComponentBase, FB_PackMLBaseModule) and your class will then have access to these properties as well.

NothingWorryGee, 2025-05-16:
Thank you very much for your response. This is my first time using GitHub, and I've just started learning SPT