Beckhoff-USA-Community/SPT-Libraries #133: Jog  has stuttery motor motion in Manual and Maintenance mode

tealex13, 2025-09-29:
[image]

In manual or maintenance mode the drive starts and stops sporadically. This does not happen in automatic mode. It also does not happen when using the MC2 library MC_Jog.

Automan-Ahmed, 2025-11-20:
[image]

Note that in Manual and Maintenance mode Jog is caled in HMICommunication method with false
So if you call jog in any other method it won't work as in HMI Communication method is called before it 
So that would create a conflict , do what we did we override HMICommunication method

tealex13, 2025-11-20:
Disabling HMI control worked! Thank you for your help.